LibreCAD/LibreCAD_3 #122: Inefficient retrival of elements within an area

gaganjyot, 2017-10-02:
From what I understand,

Lets say there is an area { 5000, 5000 } , { -5000, -5000 } area1 ( area of quadtree )

If I want to retrieve elements in area  { 100, 100 }, { 200, 200 } called area2

it checks if the area2 is in the area1 and returns all the elemets so present in the area1

I don't have a *nix platform to unittest and make sure about it. but if the understanding is correct it will be retrieving lots of entities that are not even the area. Not sure if painter will paint all the entities or only a few

https://github.com/LibreCAD/LibreCAD_3/blob/8f246bbb6b2d210b6353d71700830138b20b7876/lckernel/cad/dochelpers/quadtree.h#L342

feragon, 2017-10-03:
That's right, this test fails: 

`
TEST(Test, Test) {
    lc::QuadTree<lc::entity::CADEntity_CSPtr> qt;
    qt.insert(std::make_shared<lc::entity::Line>(lc::geo::Coordinate(-5000, -5000),
                                                 lc::geo::Coordinate(5000, 5000),
                                                 std::make_shared<lc::Layer>())
    );

    ASSERT_EQ(0, qt.retrieve(lc::geo::Area(lc::geo::Coordinate(1, -1), lc::geo::Coordinate(5000, -5000))).size()); //Returns 1 element
}
`

rvt, 2017-10-03:
In context of quad tree this is correct. The quad tree doesn't care about exact coordinates but it just finds the area's that include (even partially) in the requested area.
Since the bounding box of the lines falls over the boundig box of the requested area the line is included.

For painting it matters such that element's really outside of a quad will not be retreived, but even taht depends.

For example if you have 10 elements in a quad tree, no matter what you do, you will always retrieve them all. If you have 1000 they will be pushed down the quad tree levels and store them at finer granularity.

This could even optmise painting during canvas moving. Because you could easely request all entities up to some level X and you will only get a maximum number of entities but nothing below that (if it applies).

Hope that makes sense...