LibreCAD/LibreCAD_3 #124: Dimension object size values change when opening previously saved file

jfirrantello, 2017-10-21:
I opened a previously saved DXF created in LibreCAD (saved and re-opened same day) and found that the sizes of a part of the dimensions objects had changed significantly (see image below). Values for items such as text height, arrow size, extension line offset, etc. had changed. I have a hunch that this may have to do with the fact that I had the General Scale set to 50 in the file when everything still was working ok, but am not sure.

**Other info:**
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 7

[image]

jfirrantello, 2017-10-21:
One other comment: It may have reset the **Main drawing unit** when this occurred, unsure.

feragon, 2017-10-21:
Hi @jfirrantello, this is the repo for the next version of LibreCAD.
You should post your issue here https://github.com/LibreCAD/LibreCAD/issues

jfirrantello, 2017-10-21:
@feragon Oops, sorry about that. Will do, thanks!

closed 2017-10-21 by feragon