LibreCAD/LibreCAD_3 #135: GSoC'2018 wiki page has links to unavaliable content

Symbian9, 2018-03-22:
I just read wiki page ...

- http://wiki.librecad.org/index.php?title=GSoC_2018

... and found that it include **reference link to webpage that not exist anymore**

- http://www.gaganjyot.com/CAD_Plugin_interface

Is it issue of @gaganjyot website or is it incorrect link? Where this page live now?

P.S.: Is it same content on Medium?

- https://medium.com/@gaganjyot/librecad-v3-b7505883a975

rvt, 2018-03-22:
The links where from a old student and are ot active anymore.

Is there any specific question you might have?

Symbian9, 2018-03-23:
> Is there any specific question you might have?

On related forum topic was link to special repo, that at this moment also unavaliable.

> If anyone is interested in standardizing the JSON format for communication between plugin and LibreCAD, that would be of very great help to me.
> 
> https://github.com/gaganjyot/cad_interface
> 
> --
> Thankyou
> Gaganjyot

Was this repo merged in `master` of LibreCAD_3 or this repo sources also lost?

feragon, 2018-03-27:
This repo was not merged in LibreCAD_3/master, and the link to Medium isn't related to this project.

You can find additional information here : http://forum.librecad.org/GSoC-17-LibreCAD-td5714756.html

closed 2018-06-25 by feragon