LibreCAD/LibreCAD_3 #192: Installation failed on Mint

Guruprasad-Rane, 2018-11-26:
[CMakeError.log](https://github.com/LibreCAD/LibreCAD_3/files/2615421/CMakeError.log)
[CMakeOutput.log](https://github.com/LibreCAD/LibreCAD_3/files/2615428/CMakeOutput.log)

Tried installing LIbrecad3 on Linux Mint 19 Cinnamon but failed. CMakeError.log CMakeOutput.log files are attached.
If possible please let me know where things are going wrong.

feragon, 2018-11-26:
Do you have the pthread library installed ?

Guruprasad-Rane, 2018-11-26:
I have libpthread-stubs0-dev installed. Is this the correct package for pthread?

feragon, 2018-11-26:
It should be. Can you give me the content of this file ? [build dir]/CMakeFiles/cmTC_8dd12.dir/link.txt 
I think CMake isn't pointing to the correct library

Guruprasad-Rane, 2018-11-26:
There is no directory named cmTC_8dd12.dir in CMakeFiles. Nothing with cmTC_*****.dir

Guruprasad-Rane, 2018-11-26:
```
$ dir -R CMakeFiles/
CMakeFiles/:
3.10.2		   CMakeError.log   CMakeTmp	       feature_tests.c
cmake.check_cache  CMakeOutput.log  feature_tests.bin  feature_tests.cxx

CMakeFiles/3.10.2:
CMakeCCompiler.cmake		   CMakeSystem.cmake
CMakeCXXCompiler.cmake		   CompilerIdC
CMakeDetermineCompilerABI_C.bin    CompilerIdCXX
CMakeDetermineCompilerABI_CXX.bin

CMakeFiles/3.10.2/CompilerIdC:
a.out  CMakeCCompilerId.c  tmp

CMakeFiles/3.10.2/CompilerIdC/tmp:

CMakeFiles/3.10.2/CompilerIdCXX:
a.out  CMakeCXXCompilerId.cpp  tmp

CMakeFiles/3.10.2/CompilerIdCXX/tmp:

CMakeFiles/CMakeTmp:

```

Guruprasad-Rane, 2018-11-27:
Yes. libboost-dev as well as libboost-all-dev are installed.

feragon, 2018-11-27:
Can you try by removing everything related to threads in CMakeFiles ? 
`set(Boost_USE_MULTITHREADED ON)`, `set(Boost_USE_MULTITHREADED ON)`, `${CMAKE_THREAD_LIBS_INIT}`, `FIND_PACKAGE ( Threads )`

Guruprasad-Rane, 2018-11-28:
Tried removing or commenting as you said but no luck.
I used cmake --debug-trycompile and all the necessary temp file are there intact.
[CMakeFiles.tar.gz](https://github.com/LibreCAD/LibreCAD_3/files/2624225/CMakeFiles.tar.gz)
Here is the complete CMakeFiles folder compressed.

Guruprasad-Rane, 2018-11-28:
I did a complete fresh git clone and it partly worked this time. 
$ ./lcUI/librecad
qt5ct: using qt5ct plugin

(librecad:26519): GLib-GObject-WARNING **: 23:34:22.539: cannot register existing type 'GdkWindow'

(librecad:26519): GLib-GObject-CRITICAL **: 23:34:22.539: g_param_spec_object: assertion 'g_type_is_a (object_type, G_TYPE_OBJECT)' failed

(librecad:26519): GLib-GObject-CRITICAL **: 23:34:22.539: validate_pspec_to_install: assertion 'G_IS_PARAM_SPEC (pspec)' failed

(librecad:26519): GLib-GObject-WARNING **: 23:34:22.554: cannot register existing type 'GdkScreen'

(librecad:26519): GLib-CRITICAL **: 23:34:22.554: g_once_init_leave: assertion 'result != 0' failed

(librecad:26519): Gdk-CRITICAL **: 23:34:22.554: IA__gdk_screen_get_default_colormap: assertion 'GDK_IS_SCREEN (screen)' failed

(librecad:26519): Gdk-CRITICAL **: 23:34:22.554: IA__gdk_colormap_get_visual: assertion 'GDK_IS_COLORMAP (colormap)' failed

feragon, 2018-11-28:
I can reproduce the bug on a fresh install.
You have to change the theme in Menu > Preferences > Qt5 Settings and change the theme.
Cleanlooks works but is ugly.

Guruprasad-Rane, 2018-11-29:
Librecad3 started with cleanlooks but looks like this.
[image]

Tried to change icons theme on QT5 settings. but not luck.

Guruprasad-Rane, 2018-11-30:
Whichever theme I use to select from QT Settings Librecad use to take bb10dark.
So I modified main.cpp
    //QApplication::setStyle(QStyleFactory::create(style.getStyle()));
    QApplication::setStyle("cleanlooks");
And it worked...
Not sure whether it is problem with LIbrecad or Cinnamon. But other QT based programs are working fine.

@feragon Thank you for the support...

feragon, 2018-11-30:
Does it works better if you totally remove QApplication::setStyle ?

Guruprasad-Rane, 2018-12-01:
Yes. works better if QApplication::setStyle is removed. Style changes in QT5 settings reflected in Librecad as well.

feragon, 2018-12-01:
Ok, then I'm going to remove this.
Thanks for the report !

closed 2018-12-01 by feragon