LibreCAD/LibreCAD_3 #193: Xref support

turtle0x1, 2018-11-29:
I originally raised this in the other repository [here](https://github.com/LibreCAD/LibreCAD/issues/1025) and I think its really important feature, I would love to roll this out as but our workflow is xref centric, will it land in version 3 ? Libredwg currently has an open issue to extract xref paths from dwg files (I am trying to implement the tool but im not sure I have the required skill set).

I assume xrefs aren't part of the dxf standard so im aware that this probably causes some compatability issues

feragon, 2018-11-30:
xref will probably be supported in LibreCAD 3, but it's not in our priority list. Can you work on this? 

LibreDWG is not used by LibreCAD because of a license incompatibility, at least with v2.

turtle0x1, 2018-11-30:
@feragon Im having trouble connecting a button click on the action bar to a dialogue ui driving me pretty crazy any examples in the  existing code i can see `luaInterface:connect(layers, "layerChanged(lc::meta::Layer_CSPtr)", colorSelect, "onLayerChanged(lc::meta::Layer_CSPtr)")` but im not sure

feragon, 2018-11-30:
I saw some occurences of xref in DXF specifications. We'll of course have format incompatibilities between DXF and DWG so don't worry for this.

I suggest you to try to read xrefs from DXF files, to get a better idea of the representation. You can find our library in the LibreCAD/libdxfrw repo. 
Here is the entry point for loading a DXF file : https://github.com/LibreCAD/libdxfrw/blob/8076034027a9c7b1e7d5a7ccac095179363ef9d2/src/libdxfrw.cpp#L64

The code you see is to connect a GUI element signal to a GUI element slot. Can you give me an example of the code you're trying to run ?

rupakbajgain, 2018-12-01:
Lua and c++ are connected in lcUI/lua/qtbridge.cpp, i think clicommand does same what you are asking for,  which can be taken as example.

rupakbajgain, 2018-12-01:
For kernel it is,  lcadluascript/lckernelbridge.cpp .

turtle0x1, 2018-12-01:
Thanks for that!

feragon, 2018-12-01:
@turtle0x1 Nice work !
In xrefoperation.lua, I think this line : ```local x = lc:XrefManager();``` should be ```local x = lc.XrefManager();```

But you don't need to create an operation class to open a window.
In this line : https://github.com/turtle0x1/LibreCAD_3/blob/8793c439c8a5bf7c6d5676518f1daa0e315aeddb/lcUILua/ui/toolbar.lua#L184 , you can replace `create_basic_operation(…)` with the code to open your dialog.

turtle0x1, 2018-12-01:
Thanks, first time using lua quite a nice language though, so i can just replace `create_basic_operation` with `lc:XrefManager`? I think I tried with the dot and it didnt work but ill take another look

feragon, 2018-12-01:
Yes, so you don't need the xrefoperation.lua file

redb-2000, 2024-12-18:
Hey guys, any progress on this at all? The way i worked on autocad was completely xref centric - and really need this feature in librecad to begin making the switch

*not a coder here

dxli, 2026-10-26:
already fixed

closed 2026-10-26 by dxli

turtle0x1, 2026-10-26:
@dxli  Fixed as in you can open a dwg with xrefs on windows and it resolves? (drawing the xref layer/s with hidden/frozen constraints) - or by some other definition?