LibreCAD/LibreCAD_3 #197: Command Line output supports Bold or underscore font?

Guruprasad-Rane, 2018-12-06:
Messages shows on command prompt support any font highlighting? like Bold or  italics or underscore?

feragon, 2018-12-06:
It doesn't support RTF out of the box.
I tested this and it worked: https://stackoverflow.com/questions/1956542/how-to-make-item-view-render-rich-html-text-in-qt

If you need it, add the delegate to ui->history in lcUI/widgets/clicommand.cpp

Guruprasad-Rane, 2018-12-07:
I will try this.

feragon, 2018-12-08:
Use this file for lcUI/widgets/clicommand.cpp: https://gist.github.com/feragon/0790b191c3cc400ba5c4ad2fba54d905 

Try to enter some HTML command to test it (like <b>command</b>)

Guruprasad-Rane, 2018-12-09:
Yes. It worked perfectly :-)

Guruprasad-Rane, 2018-12-11:
Couple of issues noticed.

- Line spacing between messages has increased considerably. Tried doc.setDefaultStyleSheet but didn't work. Also tried doc.setDocumentMargin(0); but no sucess.

- Scrollbar get added after few messages and it does not auto scroll. I guess this is because of increased line spacing only.

feragon, 2018-12-11:
The auto scroll wasn't properly before. You have to scroll down one or two times to enable it.

Can you try to replace the HTMLDelegate sizeHint method with this :
```
QSize HTMLDelegate::sizeHint(const QStyleOptionViewItem& option, const QModelIndex& index) const {
    QTextDocument doc;
    doc.setHtml(option.text);
    doc.setTextWidth(option.rect.width());
    return {(int) doc.textWidth(), (int) doc.size().height()};
}
```

Guruprasad-Rane, 2018-12-11:
Before changing QSize HTMLDelegate::sizeHint
[image]

After changing QSize HTMLDelegate::sizeHint "Provide Center Point" is almost hidden because of overlap.
[image]

To reduce the line spacing I have tried
```
message("<b>CIRCLE</b><br>Provide Center Point:", self.target_widget)
```

feragon, 2018-12-11:
Did you tried without the `<br>` ? You can call `message` two times (one call per line)

I'll look at changing sizeHint method to support multi-lines items.

Guruprasad-Rane, 2018-12-13:
With out <br> with separate `message` calls
[image]

Guruprasad-Rane, 2018-12-14:
I tried changing ui->history from QListWidget to QTextEdit in readonly mode and I think it works better. Scrollbar appears but it can be made to autoscroll to the last line. Also history need not be trimmed.
[image]

Guruprasad-Rane, 2018-12-14:
`history = new QTextEdit(dockWidgetContents);`

```
void CliCommand::write(const QString& message) {
    ui->history->setHtml(ui->history->toHtml() + message);
    ui->history->verticalScrollBar()->setValue(ui->history->verticalScrollBar()->maximum());
}
```

Guruprasad-Rane, 2018-12-15:
No need for HTML delegate with QTextEdit, it can use toHtml()

Guruprasad-Rane, 2018-12-22:
Key Up and Key Down event for` CliCommand::onKeyPressed(QKeyEvent *event)` not working.
It has a condition` if(_historyIndex + 1 < _history.size())` but _historyIndex is always -1 and _history.size() is always 0
What is _historyIndex  and _history.size() ? And why this condition?

feragon, 2018-12-22:
The up and down keys allow to move into the last typed commands.
`_history` should contains every typed command, and `_historyIndex` is the current position in the history.
If `_history.size()` is always 0, that means the commands are not added to the history anymore.

closed 2020-02-16 by feragon