LibreCAD/LibreCAD_3 #202: Cancelling the ongoing command using ESC key.

Guruprasad-Rane, 2018-12-22:
It is necessary to cancel the ongoing command using ESC key.
```
void CliCommand::onKeyPressed(QKeyEvent *event) {
    switch(event->key()) {
        case Qt::Key_Up:

            if(_historyIndex + 1 < _history.size()) {
                _historyIndex++;
                ui->command->setText(_history[_historyIndex]);
            }
            break;

        case Qt::Key_Down:
            if(_historyIndex > 0) {
                _historyIndex--;
                ui->command->setText(_history[_historyIndex]);
            }
            else {
                _historyIndex = -1;
                ui->command->clear();
            }
            break;
        case Qt::Key_Escape:
            /*TODO: Code to terminate ongoing command*/
            break;

        default:
            ui->command->event(event);
            break;
    }
}
```
Is there any function available to terminate the ongoing command?

rupakbajgain, 2018-12-23:
The cancel can be done in lua by

finish_operation(id)

Where id is id of cadmdichild. 
For now it is widget itself.

rupakbajgain, 2018-12-23:
It is defined in operations.lua .

closed 2019-08-27 by feragon