LibreCAD/LibreCAD_3 #205: Perpendicular and Intersection of a vector.

Guruprasad-Rane, 2018-12-28:
My aim is to find perpendicular of a 2D vector. Which is the appropriate function?

Also wanted to find intersection of 2 vectors.

feragon, 2018-12-29:
There is no function in `lc::geo::Vector`, you have to create a `lc::entity:Line` and use the rotate function (with an angle of `M_PI / 2` or `-M_PI / 2`).

To get the intersection of two vectors, look at `lc::maths::Intersect` 
https://github.com/LibreCAD/LibreCAD_3/blob/master/lckernel/cad/math/intersect.h
The code will look like this : 
```cpp
lc::maths::Intersect intersect(lc::maths::Intersect::OnEntity, LCTOLERANCE);
intersect(v1, v2);
intersect.result(); //Contains the intersection point if it exists
```

Guruprasad-Rane, 2018-12-31:
I am trying to create circle using 3points in circleoperations.lua
I need to find the circumcenter of the triangle from 3 points 
```
function CircleOperations:Circumcenter(Point1,Point2,Point3)
    MidPoint1=Point1:mid(Point2)
    MidPoint2=Point2:mid(Point3)
    Line1=lc.entity.Line(Point1,Point2)
    Line2=lc.entity.Line(Point2,Point3)
    Line1=Line1:rotate(MidPoint1,M_PI / 2)
    Line2=Line2:rotate(MidPoint2,M_PI / 2)
end
```
I don't know how to correctly code rotate function and then get intersection of both lines in LUA

feragon, 2019-01-01:
You have to replace `M_PI` with `math.pi` and it should work

Guruprasad-Rane, 2019-01-01:
Anyways I found a direct formula for Circumcenter which is better than finding intersection of perpendicular. New code is in #206

closed 2019-01-01 by feragon