LibreCAD/LibreCAD_3 #207: Switch to LibreDWG

XVilka, 2019-01-03:
New version of LibreDWG library was released approximately month ago with a few good features
https://github.com/LibreDWG/libredwg/releases/tag/0.7

Your own library didn't update for years, feature difference is huge: https://github.com/LibreCAD/LibreCAD/tree/master/libraries/libdxfrw/src
It would be very beneficial to switch to this library finally.

new API:
  * add dwg_get_OBJECT(dwg) API for all objects in a DWG.
    analog to dwg_get_ENTITY, but including all objects, tables,
    block headers, even if unowned.
  Note that the dwg_api will be revamped from static to dynamic before 1.0.
  field accessors will be by name argument and ... (va_args)
  See the work/dynapi branch #59 (in progress).
  dwg_get_OBJECT(dwg) will be renamed to dwg_getall_OBJECT(dwg),
  dwg_get_ENTITY(blkhdr) to dwg_getall_ENTITY(blkhdr).

API breaking changes:
  * Changed BITCODE_RC from signed to unsigned char (#44)
  * LEADER changes: rename hooklineonxdir to hookline_dir,
    rename unknown_bit_4 to hookline_on,
    endptproj only until r2007. (but still not 100% correct)
  * MULTILEADER changes: rename num_vertex to numpts (analog to LEADER),
    change vertex to points[] (PR #49 Denis Pryt)
  * Fixed dwg_get_ENTITY(hdr): NULL-terminate the result (#60)

Important bugfixes:
  * Fixed all memory leaks and double-free's (#43, #42, PR #58 Denis Pryt)
  * Fixed EED sizes (PR #48, Denis Pryt)
  * Fixed BLOCK_HEADER.num_inserts, esp. with xref's (#44).
    For decode and encode.
  * Fixed LTYPE.dash (#44)
  * Fixup LEADER DXF values and path_type with annot_type mixup.
  * Fixed TABLE.value.data_type with flags kGeneral (#54 Denis Pryt)
  * Fixes VECTOR_CHKCOUNT segfault on unassigned obj->dxfname's
    (esp. for <=R12), and fix overflows (#56)
  * dwglayers: fix for DICTIONARY breaking the list output (#51)
  * dxf: fix dxf_blocks_write for empty paper_space (#52)
  * Fixed example/load_dwg.py allowing non-critical error codes

Other newsworthy changes:
  * Check for RUNNING_ON_VALGRIND and call dwg_free then, even
    if previously skipped.
  * Renamed all remaining API "null malloc" error messages to
    "Out of memory"
* Harmonized in_dxfb with in_dxf, but still WIP

rvt, 2019-01-03:
LibreDWG's support has been very spotty, how is your fork ( is it a fork??) going to change this?

lordofbikes, 2019-01-04:
You missed to copy this from http://git.savannah.gnu.org/cgit/libredwg.git/tree/NEWS:
> LibreDWG version 0.7 - released 2018/12/6 - **still alpha**:

After years of silence LibreDWG has got some attention for half a year now and its releases are marked as **still alpha**.

XVilka, 2019-01-04:
Yes, but compare the code with yours - at least it is active and has more
features.

XVilka, 2019-01-04:
@rvt it is not my fork, it is a mainline.

lordofbikes, 2019-01-04:
The point in alpha state code is:

> The API is usable but still pretty rough, we might still rename
some functions or names for better consistency and add more encoding API.

So it's just the wrong time now to decide about this step.

Symbian9, 2019-12-27:
> > LibreDWG version 0.7 - released 2018/12/6 - **still alpha**:
>
> After years of silence LibreDWG has got some attention for half a year now and its releases are marked as **still alpha**.

Propose reactivate this discussion, as there are many changes since *LibreDWG* `v0.7` release (latest stable version now is `v0.9.3` marked as **beta**) 

- https://github.com/LibreDWG/libredwg/blob/master/NEWS

<details>
  <summary>Changes since *LibreDWG* `v0.7`</summary>

```

GNU LibreDWG NEWS -- history of user-visible changes.  -*-indented-text-*-
Copyright (C) 2019 Free Software Foundation, Inc.
See the end for copying conditions.

LibreDWG version 0.9.3 - released 2019/12/25 - beta:

Minor bugfixes:
  * Added several overflow checks from fuzzing (#176)
  * Enabled AX_RESTRICT, undefining restrict in broken compilers (#141)
  * Fixed AX_ADD_FORTIFY_SOURCE warnings with gcc-9 -Wcpp
  * Disallow illegal SPLINE scenarios (#176)

LibreDWG version 0.9.2 - released 2019/11/08 - beta:

new API:
  * Added the -x,--extnames option to dwglayers for r13-r14 DWGs,
    displaying the extended variants with spaces. Since r2000 only
    those "extended" names are stored. (#167)

Minor bugfixes:
  * Fixed some leaks: SORTENTSTABLE, PROXY_ENTITY.ownerhandle for r13.
  * Add DICTIONARY.itemhandles[] for r13 and r14
  * Fixed some dwglayers null pointer derefs, and flush its output
    for each layer. (#166)

Other newsworthy changes:
  * Add geom utils to some programs: dwg2SVG and dwg2ps to transcode
    points via its OCS (i.e. extrusion vector)
  * Add basic POLYLINE_2D and LWPOLYLINE support to dwg2SVG. no arcs (bulges),
    no other curve_types (spline, bezier), no widths.

LibreDWG version 0.9.1 - released 2019/10/28 - beta:

Minor bugfixes:
  * Fixed bit_read_4BITS, off-by-one for the first bit.
  * Fixed more null pointer dereferences, overflows, hangs and memory leaks
    for fuzzed (i.e. illegal) dwg's.
  * Fixed some more compiler warnings.
  * Workaround stricter CPU's with ALIGNED_ACCESS_REQUIRED.
  * Add -fstack-protector for newer mingw, to link -lssp, which is now mandatory
    for _FORTIFY_SOURCE (a mingw-w64 libc headers bug).
  * Do not use gcc-9.0 - gcc-9.2! This series is seriously broken, not
    just for this project.
  * dwg2SVG: honor TTF font styles => Verdana or Courier (#156)

LibreDWG version 0.9 - released 2019/10/6 - beta:

new API:
  * Added the dxf importer, using the new dynapi and the r2000 encoder.
    Status: experimental. Only for r2000 dxf's yet.
  * Added utf8text conversion functions to the dynapi:
    dwg_dynapi_header_utf8text, dwg_dynapi_entity_utf8text,
    dwg_dynapi_common_utf8text converting unicode strings to UTF-8.
    And it's higher level variants: dwg_{g,s}et_HEADER_utf8text,
    dwg_{g,s}et_{ENTITY,OBJECT}_common,
    dwg_{g,s}et_{ENTITY,OBJECT}_common_utf8text,
    dwg_ent_{g,s}et_UTF8.
[... 150 lines cut ...]
    (POLYLINE, INSERT) (#83). In DWG they have different orderings,
    fixup some off-by-one decoding errors (e.g. wrong handle stream offset).
  * Fixed HATCH gradientfill color (#89)
  * decode: set proper ->dxfname from classes.inc, Fixes ACDBDICTIONARYWDFLT
    DXF object and simplifies out_dxf.
  * Fix some CRC checks: RSx vs RS_LE, add unused crc64 code.
  * More overflow and free protections added by Denis Pruchkovsky
  * Extend read/write bit_MC/UMC from 4 to 5 bytes (denis #70, PR #65)
    This is broken in all other dwg libraries.
  * Fix bit_convert_TU for >U+800 (#80)
  * Fixed more memory leaks (denis and rurban)
  * Fixed some double-free's and oob errors in EED (#99, #104)

Other newsworthy changes:
  * add --disable-bindings configure switch (#86)
  * add more codepages to dxf: ANSI_1253, ANSI_1254, ANSI_936, ANSI_949
  * add cirrus FreeBSD smoker
  * rename dxf_example.{lsp,sh} to dxf-allcvt.*
  * add dxf-roundtrip.sh, via TeighaFileConverter

LibreDWG version 0.7 - released 2018/12/6 - still alpha:
...
```

</details>

rvt, 2019-12-27:
We need to think if GPLv3 is compatible with out licence model if we want to use it.

Symbian9, 2019-12-27:
> We need to think if GPLv3 is compatible with out licence model if we want to use it.

As I understand, LibreDWG is under `GPLv3+` (NOT `GPLv3-only`)

- https://github.com/LibreDWG/libredwg/blob/master/COPYING#L637

```
   This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.
```

And *LibreCAD* `3.x` has same `GPLv3+` license

- https://github.com/LibreCAD/LibreCAD_3/blob/master/LICENSE#L10

```
This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.
```

So, there NO any issues.

rvt, 2019-12-27:
Ok, somebody overwrite the licence to be GPLv3, there was never permission to do so : See original  https://github.com/LibreCAD/LibreCAD_3/blob/645ac3d971a8c3727de206e16ed110cff6245dfe/LICENSE

I will discuss this because it's been so long ago already, but we never noticed

XVilka, 2019-12-28:
Isn't this just makes all new code (that was written after that change) to be under the GPLv3+? Since everyone who contributed after it thought the license is GPLv3?

rvt, 2019-12-28:
@XVilka yeaaa, that's a big issue indeed.
I just need to wrap around the implications about what happened.

GPLv3 limites the usage of the source code a lot, and that was never the intention.

TNick, 2019-12-28:
@gaganjyot thoughts?

liufeijin, 2020-06-08:
hope can change to libreDWG

zhengf312, 2020-06-20:
hope can change to libreDWG,too

gaganjyot, 2020-06-20:
**This message is regarding licensing. Regarding usage of LibreDWG, Please read my next message**

I investigated a lot on this during initial work and there are couple of things I'd like to mention:

And before anything else, I'd like to mention I am not a license expert/Lawyer but I've did from my understanding of licenses.

If you read the License Document it states

https://github.com/LibreCAD/LibreCAD_3/blob/master/LICENSE

> LibreCAD 3 consists of several modules and these modules may have
> a different license than the application itself. 
> For specific licenses, check respective folders.
> LibreCAD 3 itself is under the GNU GPLv3 or later License.

The current license of the LibreCAD_3 was made to GPL, but if you see, we just ship the application under GPL. That doesn't imply that the whole code needs to be GPL.

If you see the librecad Kernel has BSD3 license

https://github.com/LibreCAD/LibreCAD_3/blob/master/lckernel/LICENSE

And this was supposed to be done for all the code that we have written as a part of LibreCAD_3, we might have missed to add that but at that time everything was a part of LCKernel and hence there was a miss.

We currently depend on multiple Libraries that might have a GPL license, Cairo, Qt, Libdxfrw ( Rallaz ) or even if we think of LibreDWG.

The plan and idea was that completely whole of code written in LibreCAD_3 ( except the main.cpp file ) can be MIT/BSD-3 license.

Since we are dependent on GPL libraries, we could just make the main file as GPL license and ship the whole distributable as GPL.

But the modules ( everything can be a shared library that links to main.cpp ) like LCKernel, LCViewer and all others will be MIT/BSD-3

The reason we are using GPL now is that we don't have all the libraries that are under MIT/BSD ( like UI, DXF/DWG read/write etc. )

But the whole code is BSD-3 so it doesn't limit the usage of sourcecode at all.

If in future we could replace all the libraries with BSD-3/MIT license based ones, If everybody is okay, we can replace the GPL 3 Main file with a MIT/BSD based license and relicense the whole librecad.
Or if the team doesn't agree, the ones who wish, can still fork it and release it as another software itself under MIT/BSD-3 license.

I hope I made things clear. Please let me know if there are any other questions.

In other words:

In all the cases **the whole librecad_3 source code written by us is BSD-3/MIT**

If we are using any of the GPL libraries like dxf/dwg or even cairo/Qt we ship the software under GPL

If we are not using any of the GPL libraries we can proceed further and ship the software under MIT/BSD-3

I feel this approach of licensing is very much flexible. But still, Please let me know if there are concerns.

The code is completely under MIT/BSD-3 and lesser limiting in terms of usage.

gaganjyot, 2020-06-20:
**This message is regarding usage of LibreDWG. Regarding Licensing, Please read my previous message**

Glad that people are looking towards usage of LibreDWG, but before we just go further an use LibreDWG:

I think many of us need to understand that if a library is not active that always doesn't means that it was abandoned or is bad to be used.

Sometimes that could also mean that the reason the author developed the library is served for the author. 
Might be serving 100% for some others, some might see it serves 50% of their requirement.

Sometimes it could also mean that it is stable enough with respect to scope of development and not much can be added into it.

For an example, Netflix's hystrix has mentioned this in their readme

https://github.com/Netflix/Hystrix/blob/master/README.md

Hystrix is no longer in active development, and is currently in maintenance mode.

Hystrix (at version 1.5.18) is stable enough to meet the needs of Netflix for our existing applications. Meanwhile, our focus has shifted towards more adaptive implementations that react to an application’s real time performance rather than pre-configured settings (for example, through adaptive concurrency limits). For the cases where something like Hystrix makes sense, we intend to continue using Hystrix for existing applications, and to leverage open and active projects like resilience4j for new internal projects. We are beginning to recommend others do the same.

Now if you feel that you need such a library, and if you don't use hystrix just because its in maintainence mode, to me it doesn't seem a real good decession.

Now coming back to usage of LibreDWG in librecad,

We need to see what are the advantages we are getting by using libreDWG instead of LibDXFrw, if the pros of libreDWG outweigh we can definitely look at it.

Personally I think for what we have developed in LibreCAD_3 yet, we can stick to LibDXFrw for the moment until we face enough issues.

* The development of LibreDWG halted since there was not much motivation to develop as there were no users
* Last time I saw, there were still memory related issues in libreDWG due to which I'd refrain using libreDWG untill I am sure of them getting fixed.
* Do we have some benchmarks/comprision on the what versions are supported by both libreDWG and LibDXFrw and in case of each version what is the percentage supported.
* One more reason I'd refrain is that LibreDWG is more of a in-memory like approach where everything is stored in memory and then we have to copy it to libreCAD_3 structures which will cause higher memory usage where as libDXFrw is more of a parser style approach where it just parses and stores nothing in the memory.

* Libdxfrw supports well enough for libreCAD2 and hence it has undergone more testing than libreDWG.
* I am not completely sure but as much I think LibDXFrw implements major part of the ODA specification and there was not much more to be developed. Again thats just my understanding and I am not sure if that's the reason.

And my final 2 cents,
* We can even contribute to development of LibDXFrw as well.
* If you're still interested, It won't be much changes to incorporate libreDWG for testing purposes. If should be an interface that needs to be implemented Ries/Fergaon can help more here regarding the interface name.

liufeijin, 2020-06-20:
As an end user, in fact,i am not care on which lib is better. I just want LibreCAD2 can read and write DWG file. If someone can fix the problem on LibreCAD2 , it will be better.

flywire, 2020-06-20:
> I just want LibreCAD2 can read and write DWG file.

Why? Freely available specialised applications do it very poorly so why should LibreCAD attempt it? It will only lead to more disappointment.

I my view dwg should be stripped out of LibreCAD and supported through plugins. It should be clear people can use plugins but they are not LibreCAD. Plugin support would rest with the plugin developer.

XVilka, 2020-06-20:
Seems active enough (also now - beta): http://git.savannah.gnu.org/cgit/libredwg.git/tree/NEWS

XVilka, 2020-06-20:
@flywire the same reason nobody would have used LibreOffice or OpenOffice without MSO formats support.

flywire, 2020-06-20:
@XVilka that is a good example but the issue is you are comparing a largely documented XML format with a largely undocumented DWG format.

Symbian9, 2020-06-21:
>  comparing a largely documented XML format with a largely undocumented DWG format.

To be honest, "DWG not fully documented yet, but in progress".

rurban, 2020-09-27:
LibreDWG maintainer here.

It is still beta, because there is still no proper CAD application usage, which I would need to fix any missing or bad API's. The gold standard are the existing VBA and partially AutoLISP API's, but there are several issues with them. Recently a new and extremely simple CAD program showed up which wants to use an API similar to AutoCAD's VBA. This is a good test case to write a proper but simple enough API to write DWG's. I wrote the DXF importer which was a good testcase, but it's way too much code. The new JSON importer uses a much simpler and more generic way to write DWG's. This still needs to abstracted into a good enough API.

The above mentioned dynapi is done, and was used for the DXF and JSON importer. Very successfully.  The python and perl bindings don't use it yet, the inwork gambas bindings (VBA alike) will use it. All future bindings will use it.

Re licensing: There will be no changes. First the project is owned and defended by the FSF. You have to remember that we are dealing with one of the worst litigious companies here, right after Oracle. They sued OpenDWG for the use of trademarked names in the code, which was needed to be able to write to this format, so that programs could read that format. Only the FSF and the GPLv3 are strong enough to withstand such existential threats.

DuncanLithgow, 2020-12-28:
There continues to be progress on LibreDWG ( https://savannah.gnu.org/news/?group=libredwg ) and LibreDWG is now supported OOTB (for 2D) in FreeCAD 0.19_pre ( https://wiki.freecadweb.org/FreeCAD_and_DWG_Import )

Would it be valuable to have a plugin to allow switching to and testing LibreDWG? If this was made as a plugin would it also work for LibreCAD2?

Remember that LibreDWG also does conversion to SVG and DXF export. So it could provide new and/or improved functions.

What would this take to get done? I we can start robust testing of LibreDWG for 2D CAD this would be great and push an active project forward faster. Over at osarch we are discussing promoting DWG work to the FSF for inclusion again on their High Priority Projects list. https://community.osarch.org/discussion/385/fsf-help-us-set-high-priorities-for-2021-send-input-by-jan-8 feel free to comment there on the general topic of DWG in FOSS.

Opinions on support for DWG differ, but an issue tracker may not be the place for that discussion.

DuncanLithgow, 2020-12-28:
related to https://github.com/LibreCAD/LibreCAD/issues/1239 'Support for DWG format' in LibreCAD2

nishishailesh, 2021-12-12:
I sm not sure if this is correct place to write my experience.
System: Debian 11
Problem: Can not open DWG files.
Answer:
Application->Graphics->ODAFileConverter did not work
On commandline it showed error: qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
I ran command: QT_DEBUG_PLUGINS=1 /usr/bin/ODAFileConverter 
It gave error: QFactoryLoader::QFactoryLoader() checking directory path "/usr/bin/ODAFileConverter_22.8.0.0/platforms" ...
So I linked as follows:
cp /usr/lib/x86_64-linux-gnu/qt5/plugins/platforms /usr/bin/ODAFileConverter_22.8.0.0 -l

Now Application->Graphics->ODAFileConverter worked
Tried to convert DWG to 2018 binary DXF, file opened but nothing seen
Tried to convert DWG to 2018 ascii DXX, drawing was seen

Symbian9, 2021-12-12:
> I am not sure if this is correct place to write my experience.
> System: Debian 11
> Problem: Can not open DWG files.

@nishishailesh, are you tried to open DWG in SolveSpace? If so, you should provide SolveSpace version info and, if possible, such DWG file for test.

DuncanLithgow, 2021-12-18:
@nishishailesh thanks for reporting the problem you're having. This is not the right place to add this issue since we're discussing pro/con for a switch to LibreDWG.

Please create a new bug report and as @Symbian9 (hi there) suggests see if it opens in SolveSpace

GJoe2, 2022-02-23:
LibreDWG have support for readind dwg R2018 files (but it only write at R2000) and it is being actively developed. It would be nice to add a new branch a test it

XVilka, 2022-02-23:
Agree, latest release as of now is 0.12.5 with many changes. They also seem to fuzz the implementation as well:
- https://github.com/LibreDWG/libredwg/releases/tag/0.12.5
- https://github.com/LibreDWG/libredwg/blob/0.12.5/NEWS

paolo-caroni, 2025-01-28:
Anh news on this topic? LibreCAD3 will finnally gain a better support for dwg?

dxli, 2025-01-28:
We do not have any active v3 developer now.

dxli, 2026-10-29:
we have dwg support from libdxfrw now, still work-in-progress

rurban, 2026-10-30:
Why closing it, when the ticket is about switching to libredwg. Which is by far better than libdxfrw