LibreCAD/LibreCAD_3 #213: Build AppImage package using Travis CI

Symbian9, 2019-01-15:
Compile this application on [Travis CI](https://travis-ci.org/) upon each `git push`, and upload an [AppImage](http://appimage.org/) to your GitHub Releases page.

Providing an [AppImage](http://appimage.org/) would have, among others, these advantages:

- Applications packaged as an AppImage can run on many distributions (including Ubuntu, Fedora, openSUSE, CentOS, elementaryOS, Linux Mint, and others)
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Works out of the box, no installation of runtimes needed
- Optional desktop integration with `appimaged`
- Optional binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can optionally GPG2-sign your AppImages (inside the file)
- Works on Live ISOs
- Can use the same AppImages when dual-booting multiple distributions
- Can be listed in the [AppImageHub](https://appimage.github.io/apps) central directory of available AppImages
- Can double as a self-extracting compressed archive with the `--appimage-extract` parameter

[Here is an overview](https://appimage.github.io/apps) of projects that are already distributing upstream-provided, official AppImages.

### TODO

1. https://github.com/appimage/appimagekit
1.1 https://github.com/probonopd/linuxdeployqt
1.2 https://github.com/probonopd/uploadtool

**PLEASE NOTE:** For this to work, you need to enable Travis CI for your repository as [described here](https://travis-ci.org/getting_started) **prior to merging this**, if you haven't already done so. Also, You need to set up `GITHUB_TOKEN` in Travis CI for this to work; please see https://github.com/probonopd/uploadtool.

If you have questions, AppImage developers (i.e. @probonopd and other) are on #AppImage on irc.freenode.net.

probonopd, 2019-01-15:
Reference:
https://github.com/LibreCAD/LibreCAD/issues/858

Let me know if you need help.

feragon, 2019-01-15:
Same answer as #208: this will be done, but not in our top-priority tasks.

allenbangai, 2019-02-15:
Hi, my name Kamadje Allen and I code in C++. I will llike to contribute to you organisation. 
Please, may I know the if contributing to this issue be considered for GSOC.? 
@feragon

feragon, 2019-02-15:
Hi @allenbangai,
This issue can be considered for GSoC, but build LibreCAD as an AppImage is too simple for GSoC. So you can either propose something better (automatic releases, etc.) or take an other task.
We can discuss this on Zulip: https://librecad.zulipchat.com/

allenbangai, 2019-02-15:
OK thanks...

allenbangai, 2019-02-15:
Thanks very much

Provenance : Courrier pour Windows 10

De : Florian Roméo
Envoyé le :vendredi 15 février 2019 18:46
À : LibreCAD/LibreCAD_3
Cc : Kamadje Allen; Mention
Objet :Re: [LibreCAD/LibreCAD_3] Build AppImage package using Travis CI(#213)

Hi @allenbangai,
This issue can be considered for GSoC, but build LibreCAD as an AppImage is too simple for GSoC. So you can either propose something better (automatic releases, etc.) or take an other task.
We can discuss this on Zulip: https://librecad.zulipchat.com/
—

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

Symbian9, 2019-08-30:
> Our CI platform is building LibreCAD 3 from the master branch

As Travis CI already configured for build binary lets include also AppImage build scripts and uploading resulted AppImage binary as `continuous`  pre-release assets:

- https://github.com/probonopd/linuxdeployqt
  - https://github.com/linuxdeploy
- https://github.com/probonopd/uploadtool

lordofbikes, 2022-03-30:
I changed the title because Travis CI has changed their plans and it is annoying and extra work for OSS projects to get free capacities there.

GitHub Actions are running for LibreCAD 2 for AppImage creation now.
There is also a possible GSoC 2022 participation for this issue (opencax/GSoC#59).