LibreCAD/LibreCAD_3 #219: cmake hits undefined reference to `pthread_create'

ag88, 2019-06-05:
it seemed there are 3 instances of the same / similar problem
* https://github.com/LibreCAD/LibreCAD_3/issues/192
* http://forum.librecad.org/compile-failed-on-Debian-jessie-td5717401.html
* http://forum.librecad.org/cmake-hits-undefined-reference-to-pthread-create-td5717442.html

ag88, 2019-06-05:
i tried going into build and run
``
cmake ../subdirectory
``
apparently lcviewernoqt, lcUILua did not produce the pthread error
for the other subdirectories, the error is produced but surprisingly a makefile is generated

but if 
``
cmake ..
``
is run, it would end with
``
-- Configuring incomplete, errors occurred!
See also "/home/andrew/src/LibreCAD_3/build/CMakeFiles/CMakeOutput.log".
See also "/home/andrew/src/LibreCAD_3/build/CMakeFiles/CMakeError.log".
``

ag88, 2019-06-05:
i think this is a bug in cmake FindThreads module but so far have not found a fix for it

lordofbikes, 2019-06-05:
cross link forum post:
http://forum.librecad.org/cmake-hits-undefined-reference-to-pthread-create-td5717442.html

ag88, 2019-06-06:
i logged the issue with cmake - kitware - the company itself 
https://gitlab.kitware.com/cmake/cmake/issues/19344

closed 2026-10-29 by dxli