LibreCAD/LibreCAD_3 #220: Python script - add new feature

EngDarkRoseAlchemist, 2019-07-23:
Please, add the feature to run Python script, in addition to LUA script.

feragon, 2019-07-29:
Python support is considered now but not in our priority list.
Is there any Python feature that you need which is not available in Lua?

nealei, 2020-02-23:
@BEINAR @davidbrenner @dronov-dmitry @dxli @EngDarkRoseAlchemist @Equoda @flywire @ghost @goatchurchprime @maqifrnswa @r-a-v-a-s @runningopenloop @Rupicapra-rupicapra @Stemby @scls19fr @tom-f-oconnell @yiyuezhuo

You have previously commented via GitHub on using python in LibreCAD. Any contribution on the current proposal to be submitted within the week would be appreciated - [view it in Zulip](https://librecad.zulipchat.com/#narrow/stream/122956-Codebase/topic/V3.20Python.20API) [(login or Signup)](https://librecad.zulipchat.com/login/).