LibreCAD/LibreCAD_3 #250: Updated build instructions for Ubuntu 18.04

python50, 2020-01-05:
### Extra required Ubuntu 18.04 packages:

These packages are required in addition to the ones provided in the README on Ubuntu 18.04

sudo apt-get install libglew-dev libglm-dev

### Extra required library linkages on Ubuntu 18.04:

On Ubuntu the Cmake script does not link with a required library by default causing linker errors. Doing the following corrects this.

Add a line containing "gdk_pixbuf-2.0" (no quotes) at line 81 of LibreCAD_3/luacmdinterface/CMakeLists.txt

This links the libgdk_pixbuf-2.0 library and avoids the errors:
	CairoStore::image_create undefined reference to `gdk_pixbuf_new_from_file"
	CairoStore::image_create undefined reference to `gdk_pixbuf_get_height"
	CairoStore::image_create undefined reference to `gdk_pixbuf_get_width"
	CairoStore::image_create undefined reference to `gdk_cairo_set_source_pixbuf"

The "Correct" way to fix this would likely make use of Cmakes find_package() mechanism 

### Explicit commands for building and installing gtest:

The instructions for building gtest were vague. These commands were used successfully on Ubuntu 18.04

mkdir ~/gtest
cd ~/gtest
cmake /usr/src/gtest
make -j4
sudo cp libgtest.a /usr/lib/
sudo cp libgtest_main.a /usr/lib/

feragon, 2020-01-08:
Thanks for the feedback!
Could you test the compilation of luacmdinterface with the `fix-250` branch? 
https://github.com/LibreCAD/LibreCAD_3/blob/fix-250/luacmdinterface/CMakeLists.txt

python50, 2020-01-18:
Sorry for my delayed response, I got sidetracked.
So I created a new virtual machine from scratch and found another unlisted dependency. I needed to install libqt5svg-dev.
The updated CMake script does not seem to be successfully linking at least one required library.

> CMakeFiles/luacmdinterface.dir/main.cpp.o: In function `CairoStore::image_create(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
main.cpp:(.text._ZN10CairoStore12image_createERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN10CairoStore12image_createERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x1d4): undefined reference to `gdk_cairo_set_source_pixbuf'
collect2: error: ld returned 1 exit status

python50, 2020-01-18:
I'm writing this out so I don't forget the command again. Apparently LibreCad3 depands on libgdk-3 or libgdk-x11-2.0 ?

Okay `$ find -iname '*.so*' -execdir nm -A -D {} \; | grep gdk_cairo_set_source_pixbuf | grep -v "no symbols" > ~/foo`
yeilds

> ./libwebkit2gtk-4.0.so.37:                 U gdk_cairo_set_source_pixbuf
./libwnck-3.so.0:                 U gdk_cairo_set_source_pixbuf
./libgtk-x11-2.0.so.0:                 U gdk_cairo_set_source_pixbuf
./libgcr-ui-3.so.1:                 U gdk_cairo_set_source_pixbuf
./libxfce4ui-1.so.0.0.0:                 U gdk_cairo_set_source_pixbuf
./libgtk-3.so:                 U gdk_cairo_set_source_pixbuf
./libxfce4ui-1.so.0:                 U gdk_cairo_set_source_pixbuf
./libxkb.so:                 U gdk_cairo_set_source_pixbuf
./libgcr-ui-3.so.1.0.0:                 U gdk_cairo_set_source_pixbuf
./libgdk-3.so.0.2200.30:000000000002d460 T gdk_cairo_set_source_pixbuf
./libgdk-x11-2.0.so.0:000000000001ebb0 T gdk_cairo_set_source_pixbuf
./libgtk-x11-2.0.so.0.2400.32:                 U gdk_cairo_set_source_pixbuf
./libwnck-1.so.22.3.31:                 U gdk_cairo_set_source_pixbuf
./libexo-1.so.0.1.0:                 U gdk_cairo_set_source_pixbuf
./libgdk-3.so:000000000002d460 T gdk_cairo_set_source_pixbuf
./libgdkmm-3.0.so.1:                 U gdk_cairo_set_source_pixbuf
./libexo-2.so.0:                 U gdk_cairo_set_source_pixbuf
./libwnck-3.so.0.3.0:                 U gdk_cairo_set_source_pixbuf
./libatrildocument.so.3:                 U gdk_cairo_set_source_pixbuf
./libgtk-3.so.0:                 U gdk_cairo_set_source_pixbuf
./libxfce4panel-1.0.so.4.0.0:                 U gdk_cairo_set_source_pixbuf
./libwnck-1.so.22:                 U gdk_cairo_set_source_pixbuf
./libpixmap.so:                 U gdk_cairo_set_source_pixbuf
./libgtk-3.so.0.2200.30:                 U gdk_cairo_set_source_pixbuf
./libwebkit2gtk-4.0.so.37.39.3:                 U gdk_cairo_set_source_pixbuf
./libgdkmm-3.0.so.1.1.0:                 U gdk_cairo_set_source_pixbuf
./libexo-2.so.0.1.0:                 U gdk_cairo_set_source_pixbuf
./libxfce4panel-1.0.so.4:                 U gdk_cairo_set_source_pixbuf
./libgdk-x11-2.0.so.0.2400.32:000000000001ebb0 T gdk_cairo_set_source_pixbuf
./libatrildocument.so.3.0.0:                 U gdk_cairo_set_source_pixbuf
./libgdk-3.so.0:000000000002d460 T gdk_cairo_set_source_pixbuf
./libexo-1.so.0:                 U gdk_cairo_set_source_pixbuf

your library is probably one of these

> ./libgdk-3.so.0:000000000002d460 T gdk_cairo_set_source_pixbuf
./libgdk-3.so:000000000002d460 T gdk_cairo_set_source_pixbuf
./libgdk-3.so.0.2200.30:000000000002d460 T gdk_cairo_set_source_pixbuf
./libgdk-x11-2.0.so.0:000000000001ebb0 T gdk_cairo_set_source_pixbuf
./libgdk-x11-2.0.so.0.2400.32:000000000001ebb0 T gdk_cairo_set_source_pixbuf

python50, 2020-01-18:
Additional hiccup: it looks like path.lua needs copied manually from ./build/bin to ./build/lcUI
 
> cannot open /home/jason/Desktop/LibreCAD_3/build/lcUI/path.lua: No such file or directory
Error opening dir: No such file or directory

It "built" using libgtk-3. I have no idea if that truly is the correct library to link with.

feragon, 2020-01-20:
- I added `libqt5svg-dev` to the dependencies list. 
- LibreCAD has optional dependencies to use Cairo painter (used by luacmdinterface). To install them: `apt-get install libcairo2-dev libpango-1.0-0 libpango1.0-dev libgtk-3-dev`. `libgtk-3` was the correct package to link with.
- CMake has changed the default output dir in the recent versions. I have set it to `BUILD_DIR/bin`, and updated the generation scripts so `path.lua` is always along the binary files.

You can test again the `fix-250` branch

python50, 2020-01-24:
I created a fresh virtual machine and went through all the steps again, downloaded the repository, and switched to the fix-250 branch. It looked like the cmake file was not supplying a flag to /usr/bin/c++ to link with libgtk-3. I didn't see any directives in the CMakeLists.txt instructing it to do so.

```
cmake ..
make VERBOSE=TRUE
...
[ 90%] Linking CXX executable ../bin/luacmdinterface
cd /home/jason/Desktop/LibreCAD_3/build/luacmdinterface && /usr/bin/cmake -E cmake_link_script CMakeFiles/luacmdinterface.dir/link.txt --verbose=TRUE
/usr/bin/c++    -rdynamic CMakeFiles/luacmdinterface.dir/main.cpp.o  -o ../bin/luacmdinterface -Wl,-rpath,/home/jason/Desktop/LibreCAD_3/build/lcadluascript:/home/jason/Desktop/LibreCAD_3/build/lcviewernoqt:/home/jason/Desktop/LibreCAD_3/build/lckernel:/home/jason/Desktop/LibreCAD_3/build/lib -lpthread -lcairo -lpango-1.0 -lpangocairo-1.0 -lpangoft2-1.0 -lboost_program_options -lboost_filesystem -lboost_system -lboost_log -lboost_date_time -lboost_thread -lboost_log_setup -lboost_regex -lboost_chrono -lboost_atomic -lpthread -lcurl -lgobject-2.0 -lglib-2.0 -llua5.2 -lm -lgdk_pixbuf-2.0 ../lcadluascript/liblcluascript.so ../lcviewernoqt/liblcviewernoqt.so -lboost_program_options -lboost_filesystem -lboost_system -lboost_log -lboost_date_time -lboost_thread -lboost_log_setup -lboost_regex -lboost_chrono -lboost_atomic -lpthread -lcurl -lgobject-2.0 -lglib-2.0 -lgdk_pixbuf-2.0 -llua5.2 -lm ../lckernel/liblckernel.so ../lib/libtinysplinecpp.so -lGL -lGLU -lGLEW -lboost_log -lboost_date_time -lboost_log_setup -lboost_system -lboost_filesystem -lboost_thread -lboost_regex -lboost_chrono -lboost_atomic -lpthread -lfreetype 
CMakeFiles/luacmdinterface.dir/main.cpp.o: In function `CairoStore::image_create(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
main.cpp:(.text._ZN10CairoStore12image_createERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN10CairoStore12image_createERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x1d4): undefined reference to `gdk_cairo_set_source_pixbuf'
collect2: error: ld returned 1 exit status
luacmdinterface/CMakeFiles/luacmdinterface.dir/build.make:150: recipe for target 'bin/luacmdinterface' failed
make[2]: *** [bin/luacmdinterface] Error 1
make[2]: Leaving directory '/home/jason/Desktop/LibreCAD_3/build'
CMakeFiles/Makefile2:1072: recipe for target 'luacmdinterface/CMakeFiles/luacmdinterface.dir/all' failed
make[1]: *** [luacmdinterface/CMakeFiles/luacmdinterface.dir/all] Error 2
make[1]: Leaving directory '/home/jason/Desktop/LibreCAD_3/build'
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

After some investigation I added the following to LibreCAD_3/luacmdinterface/CMakeLists.txt
```
#GTK
find_package(GTK REQUIRED)
include_directories(${GTK_INCLUDE_DIRS})
```
and it recompiled correctly, linking the required libgtk-3.0 library

There are also a few other points of note:
```
sudo apt-get install qttools5-dev qttools5-dev-tools libqt5opengl5-dev liblua5.2-dev git g++ gcc-4.8 libcairo2-dev libpango-1.0-0 libpango1.0-dev libboost-all-dev libqt5svg5 libgtest-dev libeigen3-dev libcurl4-gnutls-dev libgtk-3-dev libglew-dev libglm-dev libqt5opengl5-dev libglew1.5-dev libqt5svg-dev  libcairo2-dev libpango-1.0-0 libpango1.0-dev libgtk-3-dev

E: Unable to locate package libqt5svg-dev`
Note, selecting 'libglew-dev' instead of 'libglew1.5-dev'
```
It looks the package name libqt5svg-dev may need to be libqt5svg5-dev to avoid an error when supplied to apt-get.

The instruction appear to omit a line saying to `cd` into `LibreCAD_3` after cloning
```
jason@jason-VirtualBox:~/Desktop$ git clone --recursive https://github.com/LibreCAD/LibreCAD_3.git
jason@jason-VirtualBox:~/Desktop$ git submodule init
fatal: not a git repository (or any of the parent directories): .git
jason@jason-VirtualBox:~/Desktop$ cd LibreCAD_3/
jason@jason-VirtualBox:~/Desktop/LibreCAD_3$ git submodule init
```
Your fix to the path.lua worked. 
Just a few more tweaks and another test and then this will be resolved.

feragon, 2020-02-01:
Thanks! The `fix-250` branch is updated with your latest comment

renegadephysicist, 2020-02-19:
I needed to install liblua5.3-dev in order for this to compile correctly

I also needed to change the following [CMakeLists.txt](https://github.com/LibreCAD/LibreCAD_3/blob/fix-250/lcviewernoqt/CMakeLists.txt#L204)

```
${OPENGL_LIBRARY}  to ${OPENGL_LIBRARIES}
${GLEW_LIBRARY} to ${GLEW_LIBRARIES}
```

System info:

Ubuntu 18.04.4 LTS
cmake version 3.16.4

feragon, 2020-02-23:
I changed `${OPENGL_LIBRARY}`  to `${OPENGL_LIBRARIES}`
`GLEW_LIBRARIES` was already used in `master` branch.

Did you have `liblua5.2-dev` installed? This version is working with Travis-CI (using Ubuntu 18.04 too).

What issue do you have with `GTK` package? LibreCAD uses GTK3, not GTK2.

I'm merging the `fix-250` branch into `master` now.

commit 48233531 (2020-02-23): Merge pull request #271 from LibreCAD/fix-250

Fix #250
closed 2020-02-23 by feragon