LibreCAD/LibreCAD_3 #255: Edit Nodes

flywire, 2020-01-20:
Edit (add, delete, move) nodes would be useful.

rupakbajgain, 2020-06-27:
Do you mean points?

flywire, 2020-06-27:
Yes

rupakbajgain, 2020-06-28:
Ok, i will try that.

rupakbajgain, 2020-07-01:
> 
> 
> Please confirm #312 functionality.
> 
> The intent of this issue is say you have a square:
> 
>     * Deleting the node at the top right corner of the square would leave the remaining nodes to form a triangle.
> 
>     * Selecting the top line and dragging a new node into it would make it five-sided, like a house with a roof. Moving to the left or right of the line would allow more points to be added.

Sorry i thought of another thing, may be this can be done in polyline only. I will think about it and try it when i find a way,

flywire, 2020-07-01:
If a square, triangle and circle are polylines then yes. Maybe a point and a line are polylines with zero and one segments respectively?

rupakbajgain, 2020-07-01:
> Please confirm #312 functionality.
>
> The intent of this issue is say you have a square:
>
>     * Deleting the node at the top right corner of the square would leave the remaining nodes to form a [...]

@flywire i am thinking of doing this my using dragpoints which are already there, how to diffrentiate between the dragging and (adding or removing node) maybe mouse button or keyboard shortcut, like del key press removes last clicked node, enter creates new node there, or diffrentiate left and right mouse click actions.
 
-> Or maybe it should be done as seperate operation. click add node and click polygon....

rupakbajgain, 2020-07-01:
> If a square, triangle and circle are polylines then yes. Maybe a point and a line are polylines with zero and one segments respectively?

Line, Square, and Triangle (arc maybe)can be represented by polyline, but not point and circle.
@feragon how should we do it.

flywire, 2020-07-01:
Dragpoints is interesting. Del deletes node, ins inserts node on line at the cursor location and the user can drag it as required. Maybe right mouse click brings up a context-sensitive menu.

Re:
- what is issue with point as zero segment polyline? It would be the same as points in the sentence above.
- circle - new node would make icecream cone shaped then the user could shape straight line(s) as required.

rupakbajgain, 2020-07-01:
@flywire ,i think insert can be done as i have in video in zulip chat,