LibreCAD/LibreCAD_3 #264: Centralization of dependencies

feragon, 2020-02-01:
LibreCAD 3 uses several dependencies. Those dependencies can be installed manually, using the system package manager or a C++ package manager.

Those dependencies are listed in multiple places (`README.md`, Travis configuration, user documentation, wiki...).
The idea is to centralize the dependencies list in one file. One advantage is to automatically update user docs when CI system warn us about changed dependencies. Those lists could be used in an install script too.

The different files could be updated in a pre-commit job

Guruprasad-Rane, 2020-04-29:
I tried installing LibreCad on fresh Ubuntu 20.04 inside a virtualbox. It gave errors.
Following packages were installed.
`sudo apt-get install git qttools5-dev qttools5-dev-tools libqt5opengl5-dev liblua5.2-dev git g++ gcc-4.8 libcairo2-dev libpango-1.0-0 libpango1.0-dev libboost-all-dev libqt5svg5 libgtest-dev libeigen3-dev libcurl4-gnutls-dev libgtk-3-dev libglew-dev libglm-dev libqt5opengl5-dev libqt5svg5-dev libcairo2-dev libpango-1.0-0 libpango1.0-dev libgtk-3-dev cmake freeglut3 freeglut3-dev mesa-utils mesa-utils-extra libbz2-dev libpthread-workqueue0 libpthread-workqueue-dev`

Is there package missing?

Here is the output in error.log

Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/vb/Desktop/LibreCAD_3/build/CMakeFiles/CMakeTmp
Run Build Command(s):/usr/bin/make cmTC_43251/fast && /usr/bin/make  -f CMakeFiles/cmTC_43251.dir/build.make CMakeFiles/cmTC_43251.dir/build
make[1]: Entering directory '/home/vb/Desktop/LibreCAD_3/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_43251.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_43251.dir/src.c.o   -c /home/vb/Desktop/LibreCAD_3/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_43251
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_43251.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD    -rdynamic CMakeFiles/cmTC_43251.dir/src.c.o  -o cmTC_43251 
CMakeFiles/cmTC_43251.dir/src.c.o: In function `main':
src.c:(.text+0x3e): undefined reference to `pthread_create'
src.c:(.text+0x4a): undefined reference to `pthread_detach'
src.c:(.text+0x56): undefined reference to `pthread_cancel'
src.c:(.text+0x67): undefined reference to `pthread_join'
src.c:(.text+0x7b): undefined reference to `pthread_atfork'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_43251.dir/build.make:106: recipe for target 'cmTC_43251' failed
make[1]: *** [cmTC_43251] Error 1
make[1]: Leaving directory '/home/vb/Desktop/LibreCAD_3/build/CMakeFiles/CMakeTmp'
Makefile:141: recipe for target 'cmTC_43251/fast' failed
make: *** [cmTC_43251/fast] Error 2
Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_cancel(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/vb/Desktop/LibreCAD_3/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_0575e/fast && /usr/bin/make  -f CMakeFiles/cmTC_0575e.dir/build.make CMakeFiles/cmTC_0575e.dir/build
make[1]: Entering directory '/home/vb/Desktop/LibreCAD_3/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_0575e.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_0575e.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.17/Modules/CheckFunctionExists.c
Linking C executable cmTC_0575e
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_0575e.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_0575e.dir/CheckFunctionExists.c.o  -o cmTC_0575e  -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_0575e.dir/build.make:106: recipe for target 'cmTC_0575e' failed
make[1]: *** [cmTC_0575e] Error 1
make[1]: Leaving directory '/home/vb/Desktop/LibreCAD_3/build/CMakeFiles/CMakeTmp'
Makefile:141: recipe for target 'cmTC_0575e/fast' failed
make: *** [cmTC_0575e/fast] Error 2`

Guruprasad-Rane, 2020-04-29:
Also terminal gave this error
CMake Error at /usr/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:164 (message):
  Could NOT find LibDxfrw (missing: LIBDXFRW_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:445 (_FPHSA_FAILURE_MESSAGE)
  persistence/cmake/FindLibDxfrw.cmake:65 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  persistence/CMakeLists.txt:24 (find_package)

Guruprasad-Rane, 2020-04-29:
LibDxfrw was compiles using  these commands
`cd libdxfrw
mkdir release
cd release
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX:PATH=/usr .. && make all
make
sudo make install`

feragon, 2020-04-29:
There was an issue with libdxfrw. Could you try again with the latest version?
```
cd libdxfrw
git pull
```

Guruprasad-Rane, 2020-04-29:
```
cd libdxfrw
mkdir release
cd release
cmake -DCMAKE_BUILD_TYPE=Release .. && make all
sudo make install
```

This code worked. Without pulling the latest code.

Guruprasad-Rane, 2020-04-29:
Installing had another new error..

```
[ 43%] Building CXX object lcadluascript/CMakeFiles/lcluascript.dir/primitive/customentity.cpp.o
/usr/bin/ld: /usr/local/lib/libdxfrw.a(drw_entities.cpp.o): relocation R_X86_64_PC32 against symbol `_ZTVSt23_Sp_counted_ptr_inplaceI12DRW_Vertex2DSaIS0_ELN9__gnu_cxx12_Lock_policyE2EE' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [persistence/CMakeFiles/persistence.dir/build.make:147: lib/libpersistence.so] Error 1

```
This is with latest pull of libdxfrw..

feragon, 2020-04-29:
What version of GCC do you have?

Guruprasad-Rane, 2020-04-29:
`gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0`
By adding
`set(CMAKE_POSITION_INDEPENDENT_CODE ON)`
in CMakeLists.txt of LibDxfrw solved this error.
And next error is again
```
[ 62%] Linking CXX executable ../bin/luacmdinterface
/usr/bin/ld: CMakeFiles/luacmdinterface.dir/main.cpp.o: in function `CairoStore::image_create(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
main.cpp:(.text._ZN10CairoStore12image_createERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN10CairoStore12image_createERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x1d7): undefined reference to `gdk_cairo_set_source_pixbuf'

```

feragon, 2020-04-29:
https://cmake.org/cmake/help/latest/prop_tgt/POSITION_INDEPENDENT_CODE.html

> This prperty is True by default for SHARED and MODULE library targets and False otherwise.

We can't force position independent code or static/shared options.

`gdk_cairo_set_source_pixbuf` is declared in [gdkcairo.h](https://gitlab.gnome.org/GNOME/gtk/-/blob/master/gdk/gdkcairo.h). The [packages](https://packages.ubuntu.com/search?suite=eoan&section=all&arch=any&keywords=gdkcairo.h&searchon=contents) containing it are:
- `libgtk2.0-dev`
- `libgtk3-dev`
Can you try again after installing `libgtk3-dev`?

Guruprasad-Rane, 2020-04-30:
I don't know much about CMAKE. Found this method somewhere on internet.

`libgtk3-dev` was not found. `libgtk-3-dev` is already installed. `gdkcairo.h` already exist at `/usr/include/gtk-3.0/gdk` 

I tried installing` libgtk2.0-dev` and reinstalling `libgtk-3-dev` but no success.

Guruprasad-Rane, 2020-05-10:
After the error "undefined reference to `gdk_cairo_set_source_pixbuf'" while running make -j 4,
running
```
cmake ..
make -j 4
```
for the second time in the same build directory compiles Librecad.
Any problem in doing this way.

Gurren-Joestar, 2020-12-10:
i get stuck on make -j 4

Gurren-Joestar, 2020-12-10:
make -j 4
make: *** No targets specified and no makefile found.  Stop.