LibreCAD/LibreCAD_3 #268: About dialog

feragon, 2020-02-17:
Containing :
- LC3 version
- Git tag
- Machine information (OpenGL...)

rupakbajgain, 2020-07-16:
i will try it now.

flywire, 2020-07-16:
Can you include OS info too?

rupakbajgain, 2020-07-17:
> Can you include OS info too?

Ok, I will try that too.

flywire, 2020-07-17:
I expect this would often be used to show user config eg in forum. It would be nice if user could copy it and paste it into forum in a collapsed state which need to be clicked to expand. That would save filling forum with seldom needed info.

rupakbajgain, 2020-07-18:
> 
> 
> I expect this would often be used to show user config eg in forum. It would be nice if user could copy it and paste it into forum in a collapsed state which needed to be clicked to expand. That would save filling forum with seldom needed info.

user config? What information should it contain, it's in html actually.

flywire, 2020-07-18:
Yes I was thinking of html. I saw it in the html specs somewhere - an element you click and it expands. (I've been looking for it again since I posted above.) It would contain all of the about information but would be pasted into the forum in closed state but could be opened by others when required.

rupakbajgain, 2020-07-18:
```spoiler Always visible heading
This text won't be visible until the user clicks.
```

rupakbajgain, 2020-07-18:
<details>
    <summary>Details</summary>
    Something small enough to escape casual notice.
</details>

flywire, 2020-07-18:
It needs a nice button name so users realise which one to copy to the forum.

Maybe  [Forum copy] or [Copy for forum].

rupakbajgain, 2020-07-18:
in github html tags work so `<details>` work fine,
but for zulip chat i have to create markdown using `spoiler` formatting help,
ok i will make copy to github or zulip.

rupakbajgain, 2020-07-23:
@flywire  thanks for the idea.

flywire, 2020-07-26:
@rupakbajgain could you put sample images and the code behind them in this issue or #324 for reference? I mean the md/html code generated not the C++ code.

rupakbajgain, 2020-07-29:
@feragon this issue should be closed, i think as i forgot to link it in my PR.

closed 2020-08-10 by feragon