LibreCAD/LibreCAD_3 #277: Build failed on openSUSE Linux: fatal error: hb.h: No such file or directory

kevinsmia1939, 2020-03-06:
I could not build LibreCAD3 on openSUSE Tumbleweed with openSUSE's Open Build Service. I have installed the following packages.
```
BuildRequires:  cmake(Qt5Concurrent)
BuildRequires:  cmake(Qt5Widgets)
BuildRequires:  cmake(Qt5UiTools)
BuildRequires:  cmake(Qt5Svg)
BuildRequires:  cmake(Qt5OpenGL)
BuildRequires:  cmake(Qt5Test)
BuildRequires:  eigen3-devel
BuildRequires:  lua53-devel
BuildRequires:  curl
BuildRequires:  boost-devel
BuildRequires:  libboost_log-devel
BuildRequires:  libboost_thread-devel
BuildRequires:  libdxfrw-devel
BuildRequires:  pkgconfig(glew)
BuildRequires:  pkgconfig(glu)
BuildRequires:  Mesa-libGL-devel
BuildRequires:  glm-devel
BuildRequires:  doxygen
BuildRequires:  freetype2-devel
BuildRequires:  cairo-devel
BuildRequires:  pango-devel
BuildRequires:  libcurl-devel
BuildRequires:  libboost_program_options-devel
BuildRequires:  libboost_system-devel
BuildRequires:  gdk-pixbuf-devel
BuildRequires:  gtk3-devel
BuildRequires:  libharfbuzz0
BuildRequires:  libharfbuzz-icu0
BuildRequires:  libharfbuzz-subset0
BuildRequires:  harfbuzz-devel
BuildRequires:  harfbuzz-tools
BuildRequires:  libharfbuzz-gobject0
BuildRequires:  typelib-1_0-HarfBuzz-0_0
BuildRequires:  libgtkmm-2_4-1
BuildRequires:  gtk2-devel
```
The error seems to be something about pango. 

```
[  133s] [ 61%] Built target lcluascript
[  133s] Scanning dependencies of target lcui_autogen
[  133s] Scanning dependencies of target librecad_autogen
[  133s] [ 61%] Automatic MOC for target librecad
[  133s] [ 61%] Automatic MOC for target lcui
[  133s] Scanning dependencies of target luacmdinterface
[  133s] [ 62%] Building CXX object luacmdinterface/CMakeFiles/luacmdinterface.dir/main.cpp.o
[  135s] In file included from /usr/include/pango-1.0/pango/pango-font.h:25,
[  135s]                  from /usr/include/pango-1.0/pango/pango-attributes.h:25,
[  135s]                  from /usr/include/pango-1.0/pango/pango.h:25,
[  135s]                  from /usr/include/pango-1.0/pango/pangocairo.h:25,
[  135s]                  from /home/abuild/rpmbuild/BUILD/LibreCAD_3-0.0+git20200229.cd24a5d1/lcviewernoqt/painters/lccairopainter.tcc:13,
[  135s]                  from /home/abuild/rpmbuild/BUILD/LibreCAD_3-0.0+git20200229.cd24a5d1/luacmdinterface/main.cpp:9:
[  135s] /usr/include/pango-1.0/pango/pango-coverage.h:28:10: fatal error: hb.h: No such file or directory
[  135s]    28 | #include <hb.h>
[  135s]       |          ^~~~~~
[  135s] compilation terminated.
[  135s] make[2]: *** [luacmdinterface/CMakeFiles/luacmdinterface.dir/build.make:63: luacmdinterface/CMakeFiles/luacmdinterface.dir/main.cpp.o] Error 1
[  135s] make[1]: *** [CMakeFiles/Makefile2:977: luacmdinterface/CMakeFiles/luacmdinterface.dir/all] Error 2
[  135s] make[1]: *** Waiting for unfinished jobs....
```
More error logs can be found here: https://build.opensuse.org/package/live_build_log/home:andythe_great/LibreCAD3/openSUSE_Tumbleweed/x86_64

Thanks.

feragon, 2020-03-08:
It seems to be a known bug: https://bugzilla.redhat.com/show_bug.cgi?id=1749943
https://bbs.archlinux.org/viewtopic.php?id=248498

As a workaround, you can disable Cairo with CMake argument `-DWITH_CAIRO=OFF -DWITH_LUACMDINTERFACE=FALSE -DWITH_RENDERING_UNITTESTS=OFF`

dabeegmon, 2020-04-20:
Added libharfbuzz-dev to a Debian testing system and was still unable to achieve a working program. Then tried disabling Cairo and was then able to get the program working. 
If desired can supply copies of resulting codes/docs in the install trial. (Please advise.)

feragon, 2020-04-23:
Fixed by #291

closed 2020-04-23 by feragon