LibreCAD/LibreCAD_3 #281: [Linux] RPMLINT warning files in /usr/local is not allowed in FHS 2.3

kevinsmia1939, 2020-03-14:
I am trying to build LibreCAD3 package for openSUSE Tumbleweed.
I use openSUSE's Open Build Service to automate this compilation.

I receive some warnings:

```
LibreCAD_3-devel.x86_64: W: static-library-without-debuginfo /usr/local/lib/libtinyspline.a
[  264s] LibreCAD_3-devel.x86_64: W: static-library-without-debuginfo /usr/local/lib/libtinysplinecpp.a
[  264s] The static library doesn't contain any debuginfo. Binaries linking against
[  264s] this static library can't be properly debugged.
[  264s] 
[  264s] LibreCAD_3-devel.x86_64: W: suse-filelist-forbidden-fhs23 /usr/local is not allowed in FHS 2.3
[  264s] LibreCAD_3.x86_64: W: suse-filelist-forbidden-fhs23 /usr/local is not allowed in FHS 2.3
[  264s] see http://www.pathname.com/fhs/ for a better location
```

I think this might be because there are .h files in usr/local/include and .so and .a in usr/local/lib.
If I have to guess, the filesystem hierarchy standard on linux prefer usr/include and usr/lib instead.
More detailed info at: https://build.opensuse.org/package/live_build_log/home:andythe_great/LibreCAD_3/openSUSE_Tumbleweed/x86_64

feragon, 2020-03-15:
Those warnings are coming from a LibreCAD3's dependency: tinyspline.
We have included it as a submodule inside the repo as it's not distributed in package managers.

The proper way is to build [tinyspline](https://github.com/msteinbeck/tinyspline) as a separate package. It would be a dependency of LibreCAD 3 package.
When building tinyspline package, you can override the installation folder: https://cmake.org/cmake/help/v3.0/variable/CMAKE_INSTALL_PREFIX.html

kevinsmia1939, 2020-03-19:
> Those warnings are coming from a LibreCAD3's dependency: tinyspline.
> We have included it as a submodule inside the repo as it's not distributed in package managers.
>
> The proper way is to build [...]

I manage to create tinyspline package and add it to the required dependecy, but how do I tell LibreCAD3 to use the installed tinyspline library and not tinyspline inside LibreCAD3?

feragon, 2020-03-19:
In this case, LibreCAD should be compiled with the installed tinyspline library, then the installed tinyspline files (`.so`s) will be taken from folders in `PATH`

msteinbeck, 2020-03-20:
@kevinsmia1939 already created an [issue](https://github.com/msteinbeck/tinyspline/issues/151) and I noted that you also discuss this problem here. Can I do something that will help you with this,e.g. creating versioned `.so` files at compile time (CMake should be able to do this)?

kevinsmia1939, 2020-03-20:
> @kevinsmia1939 created an [issue](https://github.com/msteinbeck/tinyspline/issues/151) at TinySpline and I noticed that you also discuss this problem here. Can I do something that will help you with this, e. g. creating versioned `.so` files at compile time (CMake should be able to do this)?

Hello, thank for the response.
I myself are clueless about cmake, I have been adopting others spec file.
I don't know that cmake could do give version number to a library, but I will ask around.

I have a build file for openSUSE here: https://build.opensuse.org/package/view_file/home:andythe_great/tinyspline/tinyspline.spec?expand=1

At the moment I think I have to add math library to cmake for tinyspline to compile. Because it is complaining about undefined reference to `sqrt' and `pow'.

Also tinyspline try to install to directory /usr/local which it was discourage so I add
-DCMAKE_INSTALL_PREFIX=%{_prefix} \
with %{_prefix} referring to /usr

feragon, 2020-03-21:
@msteinbeck 
I checked how to do it with CMake. 
Every `.so` file contains a [soname](https://en.wikipedia.org/wiki/Soname). You can see it as an identifier of the library. That's what's your executable will be looking for, and can be used to ensure backwards-compatibility if you set a soname with the version: `libtinysplinecpp.so.1`.
In CMake, you only have to add the version in every `set_target_properties` instructions:
```cmake
set_target_properties(tinysplinecpp_shared PROPERTIES
	OUTPUT_NAME "tinysplinecpp"
	DEBUG_POSTFIX "${TINYSPLINE_DEBUG_POSTFIX}"
	COMPILE_FLAGS "${TINYSPLINE_LIBRARY_CXX_FLAGS}"
	RUNTIME_OUTPUT_DIRECTORY ${TINYSPLINE_LIBRARY_OUTPUT_DIRECTORY}
	VERSION 1
)
```
This will set the correct soname (here `libtinysplinecpp.so.1`), and create two files: 
- `libtinysplinecpp.so.1` the original library
- `libtinysplinecpp.so` a symlink of `libtinysplinecpp.so`

feragon, 2020-03-21:
You can check how I created 3 `.so` files for libdxfrw using CMake here:
https://github.com/LibreCAD/libdxfrw/commit/b2f66370493a187bd73eb730a15bf88ed74b6cf8

kevinsmia1939, 2020-11-22:
Hi, I manage to package latest tinyspline 0.3.0, and I include it in librecad3 build dependency. However, librecad tried to compile the bundle librecad in the source file. Do you know how to make it use system install tinyspline?
Thanks.

https://build.opensuse.org/package/show/home:andythe_great/LibreCAD_3

kevinsmia1939, 2020-11-23:
Close the issue since the problem for this is gone.

closed 2020-11-23 by kevinsmia1939