LibreCAD/LibreCAD_3 #293: Update tinyspline includes

feragon, 2020-04-23:
Tinyspline header has changed from `tinysplinecpp.h` to `tinysplinecxx.h`

Guruprasad-Rane, 2020-04-28:
**While stalling Librecad** 
fatal error: tinysplinecpp.h: No such file or directory
 #include <tinysplinecpp.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
lckernel/CMakeFiles/lckernel.dir/build.make:324: recipe for target 'lckernel/CMakeFiles/lckernel.dir/cad/geometry/geospline.cpp.o' failed
make[2]: *** [lckernel/CMakeFiles/lckernel.dir/cad/geometry/geospline.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:434: recipe for target 'lckernel/CMakeFiles/lckernel.dir/all' failed
make[1]: *** [lckernel/CMakeFiles/lckernel.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
**How to bypass this?**

feragon, 2020-04-28:
The workaround is : 
- first go to tinyspline folder : `cd lckernel/tinyspline`
and go back to a lower version : `git checkout f57d124a`

Guruprasad-Rane, 2020-04-28:
tinysplinecpp.h resolved.. But another error... Not able to resolve...

[ 61%] Linking CXX shared library ../lib/liblcui.so
CMakeFiles/luacmdinterface.dir/main.cpp.o: In function `CairoStore::image_create(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
main.cpp:(.text._ZN10CairoStore12image_createERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN10CairoStore12image_createERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x1d4): undefined reference to `gdk_cairo_set_source_pixbuf'
collect2: error: ld returned 1 exit status
luacmdinterface/CMakeFiles/luacmdinterface.dir/build.make:160: recipe for target 'bin/luacmdinterface' failed
make[2]: *** [bin/luacmdinterface] Error 1
CMakeFiles/Makefile2:996: recipe for target 'luacmdinterface/CMakeFiles/luacmdinterface.dir/all' failed
make[1]: *** [luacmdinterface/CMakeFiles/luacmdinterface.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 75%] Built target lcui
[ 89%] Built target librecad
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

feragon, 2020-04-28:
Do you have those packages installed: 
- cairo
- gdk
- gdk-pixbuf ?

I'll check if there wasn't any change in Cairo packaging. As a workaround, you can add `-DWITH_LUACMDINTERFACE=FALSE` in CMake options.

Guruprasad-Rane, 2020-04-29:
It worked somehow... I don't know which package did the trick... Just before that I had installed mesa-utils-extra and cmake 3.17 from source. Cairo & GDK were already there..
I think it is important to resolve these dependencies issue.

feragon, 2020-04-29:
Fixed by #294

closed 2020-04-29 by feragon