LibreCAD/LibreCAD_3 #329: Dragging arc is crashing program

rupakbajgain, 2020-07-26:
Use drag points in arc throws const char *

rupakbajgain, 2020-07-26:
It is happening only for entity created using builder but it does not happen i create by spliting circle,
Throws error Points describe a volume, not a area.

Jedi18, 2020-07-28:
The issue seems to be in the setDragPoints function in arc.cpp in primitives. The new entity is not created properly (has inf and nan as the values), which causes the crash later on.

Jedi18, 2020-07-28:
```
    auto isCCW = bulge>0.;
    auto delta = atan(bulge) * 4.0;

    auto middle = p1.mid(p2);
    auto dist = p1.distanceTo(p2)/2.0;

    auto radius = std::abs(dist / std::sin(delta/2.0));
```
Bulge is 0, so delta is 0 since tan(0)=0, this leads to sin(0)=0, which results in radius getting infinity as it's value.

Jedi18, 2020-07-28:
We are passing in CCW() instead of bulge() as the third parameter to createArcBulge, which is the bug here.

rupakbajgain, 2020-07-28:
thanks @Jedi18 , i fixed it, have you done it too?

rupakbajgain, 2020-07-28:
Ok,  i will remove that from my code to not create conflict.

Jedi18, 2020-07-28:
That wont be a problem, it wont create a conflict if it's the same changes to that file.

Jedi18, 2020-08-02:
@feragon This issue can be closed, it was fixed in #333

closed 2020-08-02 by rupakbajgain