LibreCAD/LibreCAD_3 #359: Error buiding librecad3

jaisekjames, 2020-09-09:
When build with Lua 5.4

error: too few arguments to function ‘int lua_resume
error: ‘LUA_ERRGCMM’ was not declared in this scope

seems to [http://www.lua.org/manual/5.4/manual.html#8.3]

Pls fix it.

feragon, 2020-09-09:
The library we use for Lua bindings (https://github.com/satoren/kaguya) doesn't support Lua 5.4. 

We can't fix those errors on our side now.

closed 2020-09-09 by feragon

flywire, 2020-09-10:
Thank you.

Updated [Required libraries](https://github.com/LibreCAD/LibreCAD_3/wiki/Building#required-libraries)
> 7) Lua >= 5.2 and <= 5.3.5

jaisekjames, 2020-09-10:
Please add an option to build using different lua version in kaguya mnetioned in readme file.

"If you don’t want to use the default (system) library, add these 3 options to the cmake command cmake -DLUA_INCLUDE_DIRS=path/to/lua/header/dir 
DLUA_LIBRARY_DIRS=/abspath/to/lua/library/dir -DLUA_LIBRARIES=lualibname"

In arch linux, default is lua 5.4.

flywire, 2020-09-10:
@jaisekjames wiki is public. Can you edit https://github.com/LibreCAD/LibreCAD_3/wiki/Building directly?