LibreCAD/LibreCAD_3 #362: Drawing a line of specific length

jonadem, 2020-10-16:
This is a feature request:
Start drawing a line, click the first point and then just move the mouse in an arbitrary direction (diagonal, orthogonal, whatever) from the first point, then writing the dimension in the command line or automatically while pulling the line - and having the line drawn in that direction and in that specific length.

As requested in the forum [here](https://forum.librecad.org/Drawing-a-line-of-specific-length-like-in-AutoCAD-td5719608.html)

pgarrity18, 2021-07-14:
I would like to second this request. Or if anyone has a good workaround / modification for this, please share. Thank you.

darkbasic, 2022-02-11:
That would be awesome.

Qbiccx, 2024-02-04:
That would make a huge difference. I would love to see this feature too.