LibreCAD/LibreCAD_3 #363: Error linking liblckernel.so

deric, 2020-10-28:
Error builing current master branch on Debian:
```
[  7%] Generating ../../lckernel/build_constants.cpp
-- Build date: 2020-10-28 10:25:43
-- Build info: branch master - commit 3949a79b
[  8%] Building CXX object lckernel/CMakeFiles/lckernel.dir/build_constants.cpp.o
[  8%] Linking CXX shared library liblckernel.so
/usr/bin/ld: cannot find -ltinysplinecxx_shared
collect2: error: ld returned 1 exit status
make[2]: *** [lckernel/CMakeFiles/lckernel.dir/build.make:1284: lckernel/liblckernel.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:411: lckernel/CMakeFiles/lckernel.dir/all] Error 2

```

deric, 2020-10-28:
I followed the build instructions with:
```
$ git submodule update --recursive --remote
Submodule path '../lckernel/tinyspline': checked out 'b27d557c458dbe9b3da8f0b9102c712a97dfe1c5'
```
after checking out older commit:
```
$ git submodule update
Submodule path '../lckernel/tinyspline': checked out '8b809ac026d542933723221f6124d15f0d4a44f5'
```
the build succeeded.

msteinbeck, 2020-11-14:
Hi,

I'm the author of TinySpline. A few days ago replaced the two CMake targets `tinysplinecxx_shared` and  `tinysplinecxx_static` with a single target (`tinysplinecxx`). Whether to compile TinySpline as shared or as static library is now configured with CMake's [BUILD_SHARED_LIBS](https://cmake.org/cmake/help/latest/variable/BUILD_SHARED_LIBS.html) variable. The default is compile a static library. Also, you can now link against TinySpline using CMake export targets: https://github.com/msteinbeck/tinyspline/blob/master/test/pkg/CMakeLists.txt#L122.

PR #366 (2020-12-19): Update tinyspline dependency
Fixes #363
closed 2020-12-19 by feragon