LibreCAD/LibreCAD_3 #364: Dark theme icons support

deric, 2020-10-28:
In order to make icons visible on dark background, it's necessary to modify code for loading icons, instead of:

```cpp
_editIcon = QIcon(":/icons/layeredit");
```
use something like:
```cpp
_editIcon = QIcon::fromTheme("layer-edit", QIcon(":/icons/layeredit.svg"))
```
In case the icon is not found in any of searched paths:
```cpp
qDebug() << QIcon::themeSearchPaths();
```
the [fallback](https://doc.qt.io/qt-5/qicon.html#fromTheme-1) (current default) will be used. Later we can add support for `QIcon::setThemeName()`.

Is it ok to submit PR with such change?

The same issue is present in [LibreCAD 2](https://github.com/LibreCAD/LibreCAD/issues/1144#issuecomment-717791258).

feragon, 2020-11-24:
Most LibreCAD's icons were not designed for a dark theme.
I agree that we should support themes. How would it be stored? In another resource (.qrc) file ?

deric, 2020-11-24:
I was looking at [KDE Breeze theme](https://github.com/KDE/breeze-icons) where they use two directories with icons and a [bash script](https://github.com/KDE/breeze-icons/blob/master/icons-dark/light2Dark) to generate dark theme icons. I've tried similar approach, and the icons are at least readable (probably not graphically ideal). It should be fairly easy to maintain, when you add new icons, the dark version could be easily generated. Or do you have better idea?

u2n, 2021-05-30:
This is still an issue;  When a dark theme is used, icons disappear into the buttons, only vaguely reappearing when hovered.

As a newcomer to LC, it's alarming to see simple fixes like this not applied after 1.5 years. Guessing there must be a reason :^)

greembow, 2021-12-10:
Any word on this? I really need dark icon support.

and3rson, 2022-04-19:
Bumping this issue. Any updates on this?-

mangelba, 2024-01-07:
I modified the icons so that they look good in the dark theme. Maybe it will be useful to someone.
https://github.com/mangelba/LibreCAD-with-dark-icons/tree/dark-icons/librecad/res/icons