LibreCAD/LibreCAD_3 #365: [openSUSE] How to compile LibreCAD3 with system installed tinyspline?

kevinsmia1939, 2020-11-23:
Hello, I have manage to package tinyspline 0.3.0 into openSUSE Tumbleweed KDE.
I would like to build LIbreCAD3 for openSUSE, however, I believe LibreCAD3 bundled with older tinyspline, which I prefer to use the system install tinyspline instead of bundled dependency.
I recalled that if system installed tinyspline exist, LibreCAD3 cmake file will use that, but from my observation it does not, because the build fail exactly how old version of tinyspline failed.
How do I tell LibreCAD3 to use system installed tinyspline when building?

Thank you.

https://build.opensuse.org/package/show/home:andythe_great/LibreCAD_3

msteinbeck, 2020-11-23:
This issue may be the result of https://github.com/LibreCAD/LibreCAD_3/issues/363#issuecomment-727173669.

kevinsmia1939, 2020-11-23:
> This issue may be the result of [#363 (comment)](https://github.com/LibreCAD/LibreCAD_3/issues/363#issuecomment-727173669).

From what I understand, there are some issue that cause tinyspline to fail to build. But I manage to build tinyspline 0.3.0 separately somehow. I just want librecad to use tinyspline that was build separately not the bundled tinyspline.

rola22, 2021-05-05:
I had same problem.
I passed this compilation step by editing `LibreCAD_3/lckernel/tinyspline/src` CMakelists.txt

I have changed line no. 872: `add_library(tinysplinecxx ${TINYSPLINE_CXX_SOURCE_FILES})`
to `add_library(tinysplinecxx **SHARED** ${TINYSPLINE_CXX_SOURCE_FILES})`