LibreCAD/LibreCAD_3 #367: Dependency rapidjson should be added to build instructions

Moult, 2020-12-08:
I was following this link https://github.com/LibreCAD/LibreCAD_3/wiki/Building

I then got a build error about rapidjson. I think `rapidjson` needs to be added as a dependency. Can someone confirm?

flywire, 2020-12-08:
OS details?

Moult, 2020-12-08:
Gentoo Linux amd64.

Jedi18, 2020-12-08:
Yes, rapidjson was recently added as a dependency. The wiki needs to be updated.

Moult, 2020-12-08:
@Jedi18 cheers - I've updated the wiki! Closing issue.

flywire, 2020-12-08:
`rapidjson` has been added before. Is it updated for all builds?

Moult, 2020-12-08:
@flywire sorry, I don't understand, did I make a mistake?

flywire, 2020-12-08:
No, someone [a dev] needs to check build instructions for all platforms.

Jedi18, 2020-12-08:
@feragon rapidjson is available on Conan. Will adding `rapidjson/1.1.0` to the conanfile.txt be enough to add it?

feragon, 2020-12-08:
@Jedi18 yes, thanks for checking !
I added it in #369

closed 2020-12-08 by Moult