LibreCAD/LibreCAD_3 #368: Is posible compile Librecad_3 using Mingw in Windows 10?

Gurren-Joestar, 2020-12-08:
I know the easy way is installing Visual Studio (>20GB) and dependencies, but i would apreciate if i can compile Librecad_3 with another methods in windows 10.

feragon, 2020-12-08:
You can try using Windows Subsystem for Linux (and following Ubuntu instructions).
I tested with MinGW/MSYS2 but LibreCAD wasn't working properly. It was 2 years ago so that could have changed.

closed 2020-12-09 by Gurren-Joestar

Gurren-Joestar, 2021-02-06:
I give up compiling on WSL , now im trying using VS, but i dont know if i had to dowload 36GB of Qt 5.14.2
[image]

Gurren-Joestar, 2021-02-06:
I installed Qt creators, but it doesnt recognize msvc, so im worried why
[image]

Gurren-Joestar, 2021-02-06:
Qt recognize msvc compiler, but it doesnt matter because i dont have the qmake.exe for that compiler. Its bad because on the instruccions (set QTDIR=C:\Qt\5.14.1\msvc2017_64) it pointed on a file i dont have
[image]
[image]