LibreCAD/LibreCAD_3 #370: make -j 4 make: *** No targets specified and no makefile found.  Stop.

Gurren-Joestar, 2020-12-10:
Hi, im trying to get LibreCAD_3 on Ubuntu WSL, i need help to compile librecad_3, i got stuck on make -j 4. 

I dowload all the packages 

`apt-get install qttools5-dev qttools5-dev-tools libqt5opengl5-dev liblua5.2-dev git g++ libcairo2-dev libpango-1.0-0 libpango1.0-dev libboost-all-dev libqt5svg5 libgtest-dev libeigen3-dev libcurl4-gnutls-dev libgtk-3-dev libglew-dev rapidjson-dev`

To compile gtest i followed a third party instrucctions: https://www.eriksmistad.no/getting-started-with-google-test-on-ubuntu/

```
sudo apt-get install libgtest-dev  #already installed lines before
sudo apt-get install cmake # install cmake
cd /usr/src/gtest
_sudo cmake CMakeLists.txt
sudo make
 
# copy or symlink libgtest.a and libgtest_main.a to your /usr/lib folder
sudo cp *.a /usr/lib
```

I noticed doing a cmake was redundant because there is one already on the folder, so i skip this line. I think the compilation was sucessful because i get these files on usr\src\googletest\googletest\lib:

- libgmock.a
- libgmock_main.a
- libgtest.a
- libgtest_main.a

I copy these files on usr/lib

Then, i compile LibDxfRW line by line in prompt:
 
```
git clone https://github.com/LibreCAD/libdxfrw
cd libdxfrw
mkdir release
cd release
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON ..
make
sudo make install
```
I dont have it any issues compiling this library.

My final step was to compile LibreCAD, so i did the same steps , i talk about it on #264 

But when i run:
`make -j 4 make`

I get:
`make: *** No targets specified and no makefile found.  Stop.`

jonadem, 2020-12-21:
Did you run `cmake ..` before ?
Did it work without issue ?
Normally the last lines should look like : 
```
-- Configuring done
-- Generating done
-- Build files have been written to: /YOUR_PATH/LibreCAD_3/build
```

tormodvolden, 2020-12-22:
Never use "sudo" for compiling. Use it only for installing packages (apt install) and in some cases for "make install" (if it fails with permission issues without it, because it installs stuff to system locations).

"make -j 4 make" is wrong, it should be "make -j 4"

The instructions on https://github.com/LibreCAD/LibreCAD_3/wiki/Building work fine, I just tested them on Ubuntu 20.04. Two packages were missing in the Debian/Ubuntu/Mint list but they have now been added.

lordofbikes, 2021-01-04:
> But when i run:
> `make -j 4 make`

The difference between build instructions and your command is obviously one **make** too much.
Run `make -j 4`, this should work.
The error is about the second **make** which is interpreted as target, for which no rule exists in the makefile.

closed 2021-04-18 by Gurren-Joestar