LibreCAD/LibreCAD_3 #372: Failed to build: error: no return statement in function returning non-void

kevinsmia1939, 2020-12-23:
I failed to build on openSUSE 15.1 15.2 and Tumbleweed due to the error shown below. I download from master branch today.

Thanks.

```
[  243s] In file included from /home/abuild/rpmbuild/BUILD/LibreCAD_3-0.0+git20201220.b245f238/lcviewernoqt/painters/createpainter.cpp:4:
[  243s] /home/abuild/rpmbuild/BUILD/LibreCAD_3-0.0+git20201220.b245f238/lcviewernoqt/painters/lccairopainter.tcc: In member function 'lc::viewer::LcPainter* LcCairoPainter<T>::getCacherpainter()':
[  243s] /home/abuild/rpmbuild/BUILD/LibreCAD_3-0.0+git20201220.b245f238/lcviewernoqt/painters/lccairopainter.tcc:500:5: error: no return statement in function returning non-void [-Werror=return-type]
[  243s]   500 |     }
[  243s]       |     ^
[  243s] /home/abuild/rpmbuild/BUILD/LibreCAD_3-0.0+git20201220.b245f238/lcviewernoqt/painters/lccairopainter.tcc: In member function 'bool LcCairoPainter<T>::isEntityCached(long unsigned int)':
[  243s] /home/abuild/rpmbuild/BUILD/LibreCAD_3-0.0+git20201220.b245f238/lcviewernoqt/painters/lccairopainter.tcc:503:5: error: no return statement in function returning non-void [-Werror=return-type]
[  243s]   503 |     }
[  243s]       |     ^
[  243s] cc1plus: some warnings being treated as errors
```

feragon, 2020-12-26:
What command have you used to compile LibreCAD? 
The defaults should not be strict on compiler warnings, neither compile the Cairo part

kevinsmia1939, 2020-12-26:
> What command have you used to compile LibreCAD?
> The defaults should not be strict on compiler warnings, neither compile the Cairo part

I use open build service (OBS) to compile librecad. 
https://build.opensuse.org/package/show/home:andythe_great/LibreCAD_3

OBS add this flag to it by default.
-Werror=return-type 

I understand that it is minor issue, but I was recommended to report this as it is a good practice to fix this issue.

closed 2026-09-16 by dxli