LibreCAD/LibreCAD_3 #398: cmake is detecting lua 5.4 instead of 5.3

GJoe2, 2022-10-08:
OS: Manjaro 
```
...
-- Found Lua 5.3 include dir: /usr/include
-- Found Lua 5.3 library: /usr/lib/liblua5.4.so
-- Found Boost: /usr/lib64/cmake/Boost-1.79.0/BoostConfig.cmake (found version "1.79.0") found components: program_options filesystem system log 
-- Configuring done
-- Generating done
```
[logs_librecad.txt](https://github.com/LibreCAD/LibreCAD_3/files/9739837/logs_librecad.txt)

I have differente lua versions, it includes lua 5.4 and 5.3
$pacman -Q lua                                                                                                             
lua 5.4.4-2
pacman -Q lua53                                                                                                              
lua53 5.3.6-1

feragon, 2022-10-11:
I updated CMake files to search for Lua 5.3 only. Could you try with the latest master commit?

GJoe2, 2022-10-16:
The issue is solved with lastest master commit:
```
***** LibreCAD lcadluascript *****
-- Found Eigen3 include dir: /usr/include/eigen3
-- Found Lua: /usr/lib/liblua5.3.so;/usr/lib/libm.so (found suitable exact version "5.3.6")
```
[logs_librecad_2.txt](https://github.com/LibreCAD/LibreCAD_3/files/9793445/logs_librecad_2.txt)

closed 2022-10-16 by GJoe2