LibreCAD/LibreCAD_3 #402: Define `main` branch as default instead of `master`

jd-imi, 2024-04-15:
At the moment, the link https://github.com/LibreCAD/LibreCAD_3 points to the `master` branch while the `main` branch is ahead of several commits.

The default branch for GitHub should be `main`.

flywire, 2024-04-15:
> The default branch for GitHub should be `main`.

Only for _new_ projects but it is crazy to have master and main.

lordofbikes, 2024-04-16:
`master` is the default branch for this repo. `main` is an experimental branch from @dxli and is probably named sub optimal.
Renaming or removing `master` affects all the forks and must not be done without further discussion.
`master` was there before these `master`/`main` discussions arise. `main` is just the new standard from Git, but nobody is forced to rename `master`.
And not renaming it is just to avoid a bunch of issues with forks and cloned repos, not any kind of statement.

closed 2024-04-16 by lordofbikes