LibreCAD/LibreCAD_3 #404: Build error

kermitdafrog8, 2024-12-29:
```
/usr/include/c++/14.2.0/bits/stl_function.h:131:12: note: declared here
  131 |     struct binary_function
      |            ^~~~~~~~~~~~~~~
/tmp/SBo/LibreCAD_3-latest/luacmdinterface/main.cpp: In function ‘int main(int, char**)’:
/tmp/SBo/LibreCAD_3-latest/luacmdinterface/main.cpp:131:36: error: ‘extension’ is not a member of ‘boost::filesystem’
  131 |         fType = boost::filesystem::extension(fOut);
      |                                    ^~~~~~~~~
```
Using cmake commands
    -DCMAKE_C_FLAGS:STRING="$SLKCFLAGS" \  $SLKCFLAGS = -O2 -fPIC
    -DCMAKE_CXX_FLAGS:STRING="$SLKCFLAGS" \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DLIB_SUFFIX=${LIBDIRSUFFIX} \
    -DWITH_UNITTESTS=OFF \
    -DWITH_DOCUMENTATION=ON \
    -DMAN_INSTALL_DIR=/usr/man \
    -DCMAKE_BUILD_TYPE=Release ..

Using boost-1.87.0
gcc (GCC) 14.2.0

dxli, 2024-12-29:
We do not have any active v3 developer to address, I guess

alexander982, 2025-01-01:
Compiled normally with boost-1.74. Boost 1.85 has deprecated api https://www.boost.org/doc/libs/1_85_0/libs/filesystem/doc/deprecated.html

kermitdafrog8, 2025-01-01:
Will boost 1.78 work or is it still too new?

alexander982, 2025-01-01:
You should try. In [conanfile.txt](https://github.com/LibreCAD/LibreCAD_3/blob/master/conanfile.txt) you can see used library versions.

closed 2026-09-12 by dxli

flywire, 2026-09-12:
Can you re-enable the continuous build?